AstroHuntsman/huntsman-pocs #11: fix autoguider

lspitler, 2017-12-19:


wtgee, 2017-12-19:
This works like the other TheSkyX hardware where we read from a directory of javascript files and send those commands. For some reason the entire directory for guider is missing. I think we added it on to the laptop we used at MQ and it never made it into repo. Assuming we can find that directory there (hopefully) shouldn't be a problem.

lspitler, 2017-12-20:
@wtgee I zipped up all of that POCS and put it into the Important docs shared folder.

wtgee, 2017-12-21:
@lspitler and I tested out the connection to the autoguider as well as it taking a basic image. This was during the day so we couldn't fully test it but it addressed this particular Issue. More specific Issues should be opened when identified.

closed 2017-12-21 by wtgee