AstroHuntsman/huntsman-pocs #15: optimal focus routine

lspitler, 2017-12-19:
Half way between fine and coarse? Or Optimal then Fine? Do coarse at beginning of night, then fine rest of night?

AnthonyHorton, 2017-12-20:
* Not convinced we need more than two autofocus types. IIRC Dragonfly had a 3rd focus mode but found they didn't need it.
* Not sure what that means. BTW the step size and sweep range of both coarse and fine focus can be adjusted in the config file, if that helps.
* Once we implement writing focus position back to the config file I think we'll only need a coarse focus following a change of filter. Already on the todo list is having fine focus automatically revert to coarse focus if it fails to find a good focus. The Observatory class supports calling autofocus on lists of specific cameras, so the state machine could call coarse autofocus for some and fine for others.
* Yes. Not sure how to go about implementing this.

lspitler, 2020-11-09:
Done

closed 2020-11-09 by lspitler