AstroHuntsman/huntsman-pocs #16: come up with twilight plan

lspitler, 2017-12-19:
- open dome at sunset if clear weather
- dome flat?
- twilight flat narrow band
- coarse focus narrow band
- twilight flat broad band
- coarse focus narrow band
- fine focus sequence?
- PSF star? photometric field?
- slew to first target

wtgee, 2017-12-20:
Note that we have cameras hard-coded right now, which will need to work with the filter management system.

https://github.com/AstroHuntsman/huntsman-pocs/blob/develop/states/huntsman/calibrating.py#L23

AnthonyHorton, 2017-12-20:
The state machine can use the cameras' `filter_type` and/or autofocus exposure time attributes to decide when in twilight to do its flats. Ultimately the filter management system should be setting both of those via the config file.

AnthonyHorton, 2017-12-20:
Also, coarse focus should hardly ever be required. Once we have found a good focus for a particular lens, filter & camera combo we should be able to use the last known good focus value as a starting point for a fine focus.

closed 2020-11-09 by lspitler