AstroHuntsman/huntsman-pocs #20: figure out why 96Gb virtual memory instances are created

lspitler, 2017-12-19:
And then stop them.

wtgee, 2017-12-19:
This is a bit long but informative: https://www.youtube.com/watch?v=twQKAoq2OPE

wtgee, 2017-12-19:
@AnthonyHorton I selectively imported modules to find out when this happens. Dug down from `gunagala.imager` to `gunagala.psf` to the real culprit, which is `from astropy.modeling.functional_models import Moffat2D`. Loading that alone in a jupyter console will immediately take that to 96G.

Virtual memory is not exactly a huge problem to have large, but 96G seems a bit ridiculous. Not entirely sure how it is impacting things.

wtgee, 2017-12-19:
Digging a little more, the base class is `astropy.modeling import Fittable2DModel`, which does same thing. So does `Fittable1DModel`.

Note this does not happen on my laptop. Might be something with our numpy install.

wtgee, 2017-12-19:
A simple `from astropy import modeling` does it on the mac mini. Will try on new mac and see if same.

wtgee, 2017-12-20:
On the Mac Pro I get the following:

```
In [1]: from astropy import modeling
/Users/huntsman/anaconda3/envs/huntsman-pocs-env/lib/python3.6/importlib/_bootstrap.py:219: RuntimeWarning: numpy.dtype size changed, may indicate binary incompatibility. Expected 96, got 88
  return f(*args, **kwds)
/Users/huntsman/anaconda3/envs/huntsman-pocs-env/lib/python3.6/importlib/_bootstrap.py:219: RuntimeWarning: numpy.dtype size changed, may indicate binary incompatibility. Expected 96, got 88
  return f(*args, **kwds)
```

AnthonyHorton, 2017-12-20:
Weird. What version of `astropy`? I can't find any existing Issues relevant to this on the `astropy` github, should probably raise one.

lspitler, 2017-12-20:
Version 2.0.3

AnthonyHorton, 2017-12-20:
OK, that's the latest. How about `numpy`?