AstroHuntsman/huntsman-pocs #22: WCS information only goes into the file for the camera marked as "primary"

lspitler, 2017-12-20:
It should go into all files.

wtgee, 2017-12-20:
We want to solve them all separately? Does this need to be done while running or can it be done during a cleanup of observations as part of going into pipeline?

lspitler, 2017-12-20:
I guess it can be done by metahunter if it is going to delay POCS, etc. Unless POCS continues while solve-field is doing its thing?

AnthonyHorton, 2017-12-20:
@wtgee Yes, they all need to be solved separately. Each camera will be slightly offset & rotated from the others, slightly different distortion, etc. & we need sub-pixel accuracy WCS to rebin all the images onto a common pixel grid to combine them.

metahunter is a reasonable place to do the non-primary imagers though. metahunter can just check whether each file has a valid WCS, & if not do a solve.

lspitler, 2020-11-09:
done?

closed 2020-11-09 by lspitler