AstroHuntsman/huntsman-pocs #23: Error closing SBIGUDrv

wtgee, 2017-12-20:
Occasionally seeing this error in jupyter-console. It's related to deleting SBIGs when you don't use them. So fire up jupyter-console. Load the obsevatory. Don't actually observe. Close jupyter-console and you see this. But only every once in a while.

```
In [4]: exit
Shutting down kernel
(huntsman-pocs-env)
[/var/huntsman/huntsman-pocs]
(15:08:11)-(huntsman)-(114)-> --- Logging error ---
Traceback (most recent call last):
  File "/Users/huntsman/anaconda3/envs/huntsman-pocs-env/lib/python3.6/logging/handlers.py", line 73, in emit
    logging.FileHandler.emit(self, record)
  File "/Users/huntsman/anaconda3/envs/huntsman-pocs-env/lib/python3.6/logging/__init__.py", line 1069, in emit
    self.stream = self._open()
  File "/Users/huntsman/anaconda3/envs/huntsman-pocs-env/lib/python3.6/logging/__init__.py", line 1059, in _open
    return open(self.baseFilename, self.mode, encoding=self.encoding)
NameError: name 'open' is not defined
Call stack:
  File "/Users/huntsman/anaconda3/envs/huntsman-pocs-env/lib/python3.6/site-packages/pocs-0.5.1-py3.6.egg/pocs/camera/sbigudrv.py", line 115, in __del__
  File "/Users/huntsman/anaconda3/envs/huntsman-pocs-env/lib/python3.6/site-packages/pocs-0.5.1-py3.6.egg/pocs/utils/logger.py", line 34, in debug
Message: 'Closing SBIGUDrv driver'
Arguments: ()
--- Logging error ---
Traceback (most recent call last):
  File "/Users/huntsman/anaconda3/envs/huntsman-pocs-env/lib/python3.6/logging/handlers.py", line 73, in emit
    logging.FileHandler.emit(self, record)
  File "/Users/huntsman/anaconda3/envs/huntsman-pocs-env/lib/python3.6/logging/__init__.py", line 1069, in emit
    self.stream = self._open()
  File "/Users/huntsman/anaconda3/envs/huntsman-pocs-env/lib/python3.6/logging/__init__.py", line 1059, in _open
    return open(self.baseFilename, self.mode, encoding=self.encoding)
NameError: name 'open' is not defined
Call stack:
  File "/Users/huntsman/anaconda3/envs/huntsman-pocs-env/lib/python3.6/site-packages/pocs-0.5.1-py3.6.egg/pocs/camera/sbigudrv.py", line 121, in __del__
  File "/Users/huntsman/anaconda3/envs/huntsman-pocs-env/lib/python3.6/site-packages/pocs-0.5.1-py3.6.egg/pocs/utils/logger.py", line 34, in debug
Message: 'Closing SBIGUDrv library'
```

AnthonyHorton, 2017-12-20:
It might be best to just ditch the `SBIGUDrv` class' destructor altogether. I was never able to get it to properly unload the shared library like I had intended.

The errors seem to simply be the result of the SBIGUDrv destructor trying to send a message to the logger after bits of the logger have already been unloaded, though. Could just remove the logging bit.

wtgee, 2017-12-20:
> The errors seem to simply be the result of the SBIGUDrv destructor trying to send a message to the logger after bits of the logger have already been unloaded, though. Could just remove the logging bit.

Yes, that was my thought. I didn't see how the logger was destroyed before the destructor was complete though but I suppose it is has some delay.

I'll leave this to you fix however you see appropriate.