AstroHuntsman/huntsman-pocs #24: fits header info crash on equinox

wtgee, 2017-12-20:
Happened right as you were pulling away Wilfred... guess it is a small fix, but I am not sure where to edit, so I'll let you handle it.

Will log out of your account too now! ;-)

```
2017-12-20 16:34:22 aaomc48as.aao.gov.au root[2177] INFO PANCHAT Got it! I'm going to check out: Achernar
2017-12-20 16:34:24 aaomc48as.aao.gov.au root[2177] INFO PANCHAT Preparing the observations for our selected target
2017-12-20 16:34:25 aaomc48as.aao.gov.au root[2177] INFO PANCHAT I'm slewing over to the coordinates to track the target.
2017-12-20 16:35:03 aaomc48as.aao.gov.au root[2177] INFO PANCHAT I'm at the target
2017-12-20 16:35:04 aaomc48as.aao.gov.au root[2177] INFO PANCHAT Let's focus the cameras!
2017-12-20 16:38:01 aaomc48as.aao.gov.au panoptes[2177] INFO Fine focus plot for camera Cam00 (83F011791) with Birger Focuser focuser written to /var/huntsman/images/focus/83F011791/20171220T053511_fine.png
2017-12-20 16:38:13 aaomc48as.aao.gov.au root[2177] INFO PANCHAT I'm exploring the universe!
Exception in thread Cam00Thread:
Traceback (most recent call last):
  File "/Users/huntsman/anaconda3/envs/huntsman-pocs-env/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/Users/huntsman/anaconda3/envs/huntsman-pocs-env/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/private/var/huntsman/POCS/pocs/camera/sbig.py", line 270, in process_exposure
    equinox = float(info['equinox'].replace('J', ''))
  File "/Users/huntsman/anaconda3/envs/huntsman-pocs-env/lib/python3.6/site-packages/astropy/time/core.py", line 1151, in __getattr__
    return self.__getattribute__(attr)
AttributeError: 'Time' object has no attribute 'replace'
```

lspitler, 2017-12-20:
Realised this issue should go into the `AstroHuntsman/POCS` repo. But I can't seem to add issues to it...?

lspitler, 2017-12-20:
Oh, and it appears the offending line creates a variable that is not used later...

wtgee, 2017-12-20:
> Will log out of your account too now! ;-)

I didn't even read the issue because I thought it was something I did so I was very confused by the comments alone. :)

wtgee, 2017-12-20:
Fix should be in https://github.com/panoptes/POCS/pull/225

wtgee, 2017-12-21:
Confirmed fix during live testing. Closing.

closed 2017-12-21 by wtgee