AstroHuntsman/huntsman-pocs #77: Fix test_huntsman tests

AnthonyHorton, 2018-10-02:
The `huntsman-pocs` test suite does not run successfully because the tests in `huntsman/tests/test_huntsman.py` have not been updated to keep track of changes in `panoptes/POCS`.  Currently the tests hang indefinitely in `test_run_wait_until_safe`.

The tests here are largely a duplicate of those in `panoptes/tests/test_pocs.py`, and they should be apart from a few parameter values. Several of the tests in `huntsman/tests/test_huntsman.py` have now diverged significantly from the `panoptes/POCS` versions, though. The troublesome `test_run_wait_until_safe` is one example. 

I've been attempting to get the test suite running in Travis CI again in #76 but it's clear that a number of tests need fixing, not just config files, so I'm opening a separate Issue for fixing tests.

AnthonyHorton, 2018-10-02:
This is a blocking issue for merging any PRs into `huntsman-pocs` as no branches are able to pass tests right now.

wtgee, 2018-10-02:
So is the primary task here to update `test_huntsman.py`? It looks like you have put through other fixes for travis (that's always a fun hellish commit and wait process).

I can start looking at the test file if that is what is needed.

AnthonyHorton, 2018-10-02:
Yes, that's right. I think Travis is fixed now, just need to update those tests. Should mostly be a case of copy-pasting current versions from POCS.

wtgee, 2018-10-02:
> Yes, that's right. I think Travis is fixed now, just need to update those tests. Should mostly be a case of copy-pasting current versions from POCS.

Yup, doing that now.

PR #79 (2018-10-02): Fixing tests
This is a mish-mash of fixes to get testing working. Includes a better (hopefully) way of loading config files. Ideally would come up with a more generic way for derivatives of POCS to do this.

Note that I've forced the tests with the mongo db only, unlike POCS, which does all three db types

Fixes #77 (still could use plenty more tests).
closed 2018-10-02 by wtgee