AstroHuntsman/huntsman-pocs #98: Figure out transfer error from Pi

wtgee, 2018-11-25:
It's unclear if there is actually a problem transferring the files or if the Pi is just dying for some reason and thus transfer doesn't complete. Since we haven't seen the Pi die at any other time besides this it is most likely related directly to the transfer.  Note that we are selectively seeing this on a Pi, not in all cases.

```bash
pyro.py:377] File transfer huntsman@10.88.21.143:/var/huntsman/temp/focus/./83F011167/* -> /var/huntsman/images/focus/ failed
```

wtgee, 2019-10-25:
@AnthonyHorton is this still an issue?

AnthonyHorton, 2019-10-25:
@wtgee Don't know. Need to spend some time running the system to see if it reoccurs. I'm expecting at least the Pi's with FLI cameras to fall over during autofocus again, at which point we'll have some data with which to investigate.

AnthonyHorton, 2020-02-07:
Closing this as moot now that we're not longer doing file transfers (using sshfs instead).

closed 2020-02-07 by AnthonyHorton