AstroHuntsman/huntsman-pocs #105: async_wait receives incorrect result from pyro camera

wtgee, 2018-12-01:
There is a problem with the the return result for the pyro cameras that manifests itself in the flat_field and focus methods in `huntsman-pocs`. It's not clear if it would also be present in the working version from POCS. It also doesn't appear to happen during normal observations.

It looks like pyro is trying to send an astropy Quantity over the wire but having problems but it was really not clear to me what the expected return value should be.

The fix to get it working was to wrap in a Try/Except block and return True anyway: https://github.com/AstroHuntsman/huntsman-pocs/blob/backup_notebook/huntsman/camera/pyro.py#L386

Example from the logs:
```
D1130 09:43:49.330 pyro.py:178] Taking 1.0 second exposure on camera.sbig.006: flat_02.fits
D1130 09:43:49.331 observatory.py:267] Waiting for flat-field image
D1130 09:43:50.336 observatory.py:267] Waiting for flat-field image
D1130 09:43:51.322 pyro.py:389] Problem in wait: unsupported serialized class: astropy.units.quantity.Quantity
```

AnthonyHorton, 2018-12-03:
Huh. Thought I'd been pretty careful to sanitise all the values that get passed through the Pyro serialiser. Will investgate.

AnthonyHorton, 2018-12-03:
@wtgee Weird. Have looked through the code and can't see how the Pyro cameras would ever be trying to send a Quantity through the serialiser. Going to have to add a bunch of extra logging to work out what is going on. Does this happen with every exposure taken with the flat field and focus methods, or is it intermittent?

wtgee, 2018-12-03:
> Does this happen with every exposure taken with the flat field and focus methods, or is it intermittent?

It would fail every time for the first image and the try/except was the only way around that. So can't say for sure it happens _every_ time but I would guess so.

wtgee, 2018-12-03:
I also wasn't watching the Pi camera logs at this time nor did I go back and look. I wasn't actually sure what value to expect. The `_async_wait` seems to just want a boolean so I was confused why it would ever be something with a `Quantity`.

lspitler, 2019-12-16:
OK, I think I found it @wtgee @AnthonyHorton . 

While running `take_evening_flats` `take_observation` gets a `exptime` that is a Quantity:
https://github.com/AstroHuntsman/huntsman-pocs/blob/develop/huntsman/observatory.py#L267
and:
https://github.com/AstroHuntsman/huntsman-pocs/blob/develop/huntsman/observatory.py#L371

While `/panoptes/POCS/camera/camera.py` does find it and sends it in as `seconds=exptime` in `self.take_exposure(`:
https://github.com/panoptes/POCS/blob/6cbe36ebc154d9c054a20efab458e0a1d022d86b/pocs/camera/camera.py#L298

...it doesn't strip `exptime` or at least the units from  `kwargs`. This trips up the `pyro.py` as it can't handle `Quantities` yet.

Solution in a new PR will be somewhat of a hack in `pyro.py` as that's the problem - normal POCS doesn't care. So I might just add:
```python
# strip units of exptime
if 'exptime' in kwarg:
    exptime = get_quantity_value(exptime, u.exptime)
```
to:
https://github.com/AstroHuntsman/huntsman-pocs/blob/d9a582ff5673caa7bb58fbe6dde242f1b9870511/huntsman/camera/pyro.py#L233

AnthonyHorton, 2019-12-16:
Root cause of the issue is in fact in POCS. 

`pocs.camera.AbstractCamera.take_observation()` takes an optional `exptime` keyword argument (in `kwargs`), which is used to override the `.exptime` attribute of the `Observation`. This keyword argument gets passed on in `kwargs` to the `._setup_observation()` method, which sets the exposure time with the line `exptime = kwargs.get('exptime', observation.exptime.value)`. This leaves `exptime` in `kwargs`, so that when `.take_exposure()` is called it also receives an `exptime` keyword argument.

This serves no purpose as `take_exposure` does not use an `exptime` argument (exposure time is set with the `seconds` argument instead), but can cause problems for the Pyro `Camera` class if `exptime` is an `astropy.units.Quantity` instead of a plain float or integer. The `.take_exposure()` method of the `huntsman-pocs.camera.pyro.Camera` class passes on `*args` and `**kwargs` in the call to the remote `huntsman-pocs.camera.pyro.CameraServer.take_exposure()` method, but we haven't told Pyro how to serialize `astropy.units.Quantity` objects so this will cause an error if there is a `Quantity` lurking in `kwargs`.

Ideally we should setup serializers for `astropy.units.Quantity`, and probably do sanitisation of `args` and `kwargs` before passing them too, but the simplest solution for this specific issue is to fix the problem at the source by removing `exptime` from `kwargs` before passing to `take_exposure`.

This can be achieved by changing https://github.com/panoptes/POCS/blob/develop/pocs/camera/camera.py#L703 to `exptime = kwargs.pop('exptime', observation.exptime.value)`.

AnthonyHorton, 2020-02-07:
Fixed by https://github.com/panoptes/POCS/pull/935

closed 2020-02-07 by AnthonyHorton