AstroHuntsman/huntsman-pocs #108: CCD burn in/persistence

fergusL, 2019-05-24:
[image]

Noticed what appears to be CCD burn in/persistence while collecting some 0~1 second darks for something. I haven't checked through the data for all the cameras, but camera 83F011791 is affected.
What seems to be happening is that we are taking flats early and getting saturated exposures, creating noticeable persistence when we create the follow up dark. The attached image shows several 1 second darks taken over the course of the evening. You can see the burn  in along the edge of the ccd, which fades as the evening progresses. 

I think you can sort of "flush out" the ccd by taking a bunch of biases before you take an actual dark/sci exposure but Anthony has noted that sometimes persistence only fades with time.

This means we either need to add a "flushing" mechanism or we need to get much better at picking when to start our flats.

fergusL, 2019-05-24:
If anyone wants to look at the data themselves, the image above is using the dark data from flat observations on the 8th of feb 2018

wtgee, 2019-05-24:
> You can see the burn in along the edge of the ccd, which fades as the evening progresses.

On what kind of time scale? Minutes or Hours?

fergusL, 2019-05-30:
> On what kind of time scale? Minutes or Hours?

The screenshot shows the dark exposures taken each time we took a saturated flat on a single night. So they were all taken in ~2 hours but sporadically spaced (whenever somebody tried to start flats again). 

I'd assume the persistence is fading quickly inbetween exposures, its just everytime you take a new saturated flat it "refreshes". The fading you see over the course of the night is perhaps due to a smaller fraction of pixels saturating with each new flat, as the twilight sky is darkening? As opposed to all the persistence coming from the initial saturated exposure and slowly fading over 2 hours.