AstroHuntsman/huntsman-pocs #113: Observatory doesn't have .take_flat_fields attribute

AnthonyHorton, 2019-10-22:
The calibrating state attempts to decide whether it should be taking twilight flats by checking the truth of `observatory.take_flat_field`. However if `take_flat_field` has not been explicitly set (somewhere in `huntsman_local.yaml`?) then it does not receive a default value and `.take_flat_field` simply does not exist, resulting in an `AttributeError`.  Should either be assigning a default value of `False` to `.take_flat_field` or catching the `AttributeError` and treating is as equivalent to `False`.

AnthonyHorton, 2019-10-22:
OK, this one was just user error.  I did `from huntsman.observatory import Observatory` instead of `from huntsman.observatory import HuntsmanObservatory`, so I was actually using the base `pocs.observatory.Observatory` class that gets imported into `huntsman.observatory` instead of `HuntsmanObservatory`. `HuntsmanObservatory` does have the `.take_flat_fields` attribute, set by a `take_flats` argument that defaults to `False`.

closed 2019-10-22 by AnthonyHorton

wtgee, 2019-10-22:
I want to say this was originally set only as a parameter passed in during `Observatory` creation but can't remember if that was on purpose or simply never implemented correctly.

Note https://github.com/panoptes/POCS/pull/729