AstroHuntsman/huntsman-pocs #120: upon failed astrometry plate solve continue observing

lspitler, 2019-10-25:
According to @wtgee current action is to shutdown and park if e.g. there are clouds on the first image of a sequence. We should continue observing I think or else trigger a "cloud" mode...?

lspitler, 2019-10-26:
@wtgee on Friday I thought you said the default observing mode will result in a shutdown if the plate solve doesn't work. @AnthonyHorton 's slack comms seems to contradict this - POCS went to the next dither position after a plate solve failure. So maybe I misunderstood what you said?

lspitler, 2019-10-31:
@wtgee just want to see if this is really an issue or not?

wtgee, 2019-10-31:
> @wtgee on Friday I thought you said the default observing mode will result in a shutdown if the plate solve doesn't work. @AnthonyHorton 's slack comms seems to contradict this - POCS went to the next dither position after a plate solve failure. So maybe I misunderstood what you said?

That is how it should work in POCS by default but we might have set it so that it doesn't need that in `huntsman-pocs`.

Let's leave open for now and I'll double-check the code and we'll see how it goes next time on sky.

AnthonyHorton, 2019-10-31:
Yup, definitely saw some plate solve failures go scrolling by in the logs, but Huntsman kept going. Possible we made `huntsman-pocs` override the normal behaviour of `POCS` at some point.

lspitler, 2019-11-05:
OK thanks @AnthonyHorton I'm closing this now. @wtgee feel free to revive.

closed 2019-11-05 by lspitler