AstroHuntsman/huntsman-pocs #185: Don't require all cameras in the ready state

wtgee, 2020-05-20:
Ideally this is where we would have the boolean option `require_all_cameras`.  In a first pass this could be done with the for loop, it just means that you would always hit the `max_attempts` if any of the cameras were taking a long time to start.  

That should then be switched somehow so the camera returns an event and this implements some other loop to wait and check.

Anyway, just marking for a future PR.

_Originally posted by @wtgee in https://github.com/AstroHuntsman/huntsman-pocs/pull/184_

danjampro, 2020-05-21:
My only worry is that a camera could die (maybe something would appear in the logs), but nobody would realise for X amount of time. This is especially true while running in normal operations when the logs will probably be checked infrequently. I think there should be some kind of notification system if some of the cameras do not respond - any ideas on this?

danjampro, 2020-05-21:
Marking as resolved following merge of 184.

closed 2020-05-21 by danjampro