AstroHuntsman/huntsman-pocs #193: Relative path in state table causes problems if POCS is run outside of huntsman-pocs directory

danjampro, 2020-08-18:
If a `POCS` instance is created outside of the `huntsman-pocs` root directory, `PanStateMachine._load_state` will raise an `InvalidConfig` error.

Relative path defined [here](https://github.com/AstroHuntsman/huntsman-pocs/blob/da67091a45f74d5760ac8cf1cb4984ef20cc8db4/resources/state_table/huntsman.yaml#L4).

Read into `PanStateMachine` [here].(https://github.com/panoptes/POCS/blob/216407bbf8b12a0d021e0688f8c9a7442cfeb0f6/src/panoptes/pocs/state/machine.py#L32)

AnthonyHorton, 2020-08-18:
The answer to the question "do we need to hard-code the <x>?" is always "No!" ;-)

Seems like the environment variables (e.g `$PANDIR`) should be used here?

wtgee, 2020-08-18:
I see the problem with this line assuming a relative path, but unfortunately @AnthonyHorton suggestion of `$PANDIR`, while normally correct, is not right here.

https://github.com/panoptes/POCS/blob/216407bbf8b12a0d021e0688f8c9a7442cfeb0f6/src/panoptes/pocs/state/machine.py#L32

This is later used by `load_module` but unfortunately **it does** have a hard-coded path, namely it assumes the eventual state is under the `panoptes` namesspace:

https://github.com/panoptes/POCS/blob/216407bbf8b12a0d021e0688f8c9a7442cfeb0f6/src/panoptes/pocs/state/machine.py#L340-L344

So what would really have to happen is to correct that hard-coded `panoptes` as the top-level namespace for the state module.

It should be pretty easy™.  :rofl:

danjampro, 2020-08-18:
Yeah I was a little worried when I saw that hard coded "panoptes", which isn't actually there in older versions of `POCS`. The version we are currently using takes the top level as "states", hence the problem with relative relative paths. 

Not sure what the best way is to solve that. Is it easy to retrieve the top-level namespace? Alternatively we could override the class in `huntsman-pocs`, but that is probably a last resort.

wtgee, 2020-08-19:
> Yeah I was a little worried when I saw that hard coded "panoptes", which isn't actually there in older versions of `POCS`. The version we are currently using takes the top level as "states", hence the problem with relative relative paths.

Yup, this was me being lazy when I introduced the namespace for the repo. Bad coder!

> 
> Not sure what the best way is to solve that. Is it easy to retrieve the top-level namespace? Alternatively we could override the class in `huntsman-pocs`, but that is probably a last resort.

PANOPTES is actually relying on that default value that is set in the `_states_location` initializer on line ~33 so it shouldn't be too bad.  I started a PR, will submit shortly.

wtgee, 2020-10-28:
@danjampro I think this has been resolved?

closed 2020-10-30 by danjampro