AstroHuntsman/huntsman-pocs #205: Camera errors not handled by Huntsman "observing" state.

danjampro, 2020-09-03:
If a timeout error (or maybe any other error) is encountered by the camera itself, the error does not get raised at the state machine level. This is potentially a fatal bug because it can cause the "observing" state to hang indefinitely.

wtgee, 2020-09-03:
Do you have anything from the logs that shows this happening?

danjampro, 2020-09-04:
@wtgee 

Camera logs:
```
D0903 10:09:15.884                 libasi.py:238] Set EXPOSURE to 300.0 s on camera 0
D0903 10:09:15.885                 libasi.py:450] Exposure started on camera 0
E0903 10:14:45.893                  error.py:016] Timeout: Timeout waiting for exposure on ZWO ASI Camera (361d420013090900) with Birger Focuser & ZWO Filter Wheel to complete
E0903 10:14:45.894                 camera.py:598] Error while waiting for exposure on ZWO ASI Camera (361d420013090900) with Birger Focuser & ZWO Filter Wheel: Timeout: Timeout waiting for exposure on ZWO ASI Camera (361d420013090900) with Birger Focuser & ZWO Filter Wheel to complete
```

And lots of 
```
Waiting for images ...
```
in the main logs.

closed 2021-07-28 by danjampro