AstroHuntsman/huntsman-pocs #218: running checklist of things to do to setup an automated pi setup with cloud init

lspitler, 2020-09-18:
Supporting #209 , these are things that the team can help @wtgee with to setup this scheme:

- [ ] zshrc file 
- [ ] other post-install commands
- [ ] setup ssh keys
- [ ] setup cron jobs (not sure what they are)
- [ ] auto mount control computer ssh share
- [ ] use shared `startup.sh` on the ssh mount to customise the startup of each pi (e.g. pull latest docker compose) https://cloudinit.readthedocs.io/en/latest/topics/examples.html#run-commands-on-first-boot
- [ ] point to new docker container?
- [ ] filling in all placeholders in current init file
- [ ] extras, e.g. set up cron jobs (downloading IERS for astroplan), auto-start camera services (?), etc.

lspitler, 2020-09-19:
[image] [setup and test cloud init](https://trello.com/c/wZKp2f9B/510-setup-and-test-cloud-init)

danjampro, 2021-01-13:
Superseded by `huntsman-pocs` wiki

closed 2021-01-13 by danjampro