AstroHuntsman/huntsman-pocs #251: Pyro objects do not unregister from nameserver

danjampro, 2020-12-08:
If a pyro device is terminated (or crashes), it should unregister itself on the name server. Additionally, when starting a new pyro service, we should unregister any pyro object with the same service name to prevent registration errors. 

This is particularly important for getting the restart on reboot working properly.

AnthonyHorton, 2020-12-08:
They used to unregister themselves, it was in the camera server (now service) `__del__()` method. It's impossible to guarantee that gets run though, as it relies on Python being allowed to clean up properly. If the OoM killer just splats the Python process or the Pi hard resets no Python code is going to be able to ensure the remote object gets unregistered.

We also set the name server to automatically unregister any remote objects that it was unable to contact for a set timeout time.

I like the idea of checking for and removing duplicates, though.

lspitler, 2020-12-08:
I thought we'd attempt to reboot a pi fast enough so that the name server doesn't know it has gone offline?

lspitler, 2020-12-09:
Relevant to: https://github.com/AstroHuntsman/huntsman-pocs/pull/243

lspitler, 2020-12-09:
@AnthonyHorton would this work:
```python

                    if self.config['pyro']['reboot_on_failure']:
                        self.logger.debug(f"Rebooting computer hosting camera {self._camera}")
                        self._camera.__del__()
                        time.sleep(5)
                        subprocess_args = ["sudo", "reboot", "now"]
```
?