AstroHuntsman/huntsman-pocs #273: Skip a filter for flats if flats were already taken with that filter

JAAlvarado-Montes, 2020-12-24:
If POCS starts twilight flats late (because of weather or having to restart POCS) it should not try to do all of the filters. It'll be too dark & it will just get further & further behind schedule as it's forced to use long exposures.

wtgee, 2020-12-24:
From @AnthonyHorton in slack

> Somewhat related to the goal I've previously discussed of having a narrow range of Sun altitudes for each filter, instead of a wide range of Sun altitudes for the twilight flats state as a whole. That would give us more consistent results, especially good for cancelling sky gradients by combining evening & morning flats, & would naturally solve the "starting late" issue. (edited)

lspitler, 2020-12-25:
Good idea @AnthonyHorton .  @fergusL can you advise on preferred altitudes, or do we need to test? Or just pick one for each filter and see if it works for the whole sequence. One idea is to take e.g. 5 second exposures in a filter through the entire twilight sun altitude sequence  and get counts vs sun altitude. Then we'll know for a filter roughly what altitude we should start.