AstroHuntsman/huntsman-pocs #290: Allow cameras to wait for config server

danjampro, 2021-01-11:
Cameras should wait for the config server to be up before trying to initialise. At the moment camera services throw errors if they are started up before the config server.

wtgee, 2021-01-11:
@danjampro the `wait-for-it` scripts can be used for this, as in:

https://github.com/panoptes/POCS/blob/e41c2b1ae34188f91b0b7963886ebd541f876ea2/docker/docker-compose.yaml#L34-L35

It currently exists in the `panoptes-utils` repo and docker image. I will be stripping it out of repo soon, but installing via the system (i.e. `apt`) so it will be available `wait-for-it` instead of a full path.

JAAlvarado-Montes, 2021-01-12:
@danjampro, wasn't this closed by #294 ?

closed 2021-01-13 by danjampro