AstroHuntsman/huntsman-pocs #301: Old logs have weird formatting

danjampro, 2021-01-13:
For some reason the old (archive / backup) logs (`panoptes_xxx.log`) do not have the same formatting as the current `panoptes.log` and are thus difficult to read.

danjampro, 2021-01-27:
@wtgee any intuition on this?

wtgee, 2021-01-27:
Sorry, I was expecting more info about the formats (I thought I made that comment initially, but I guess not).

The formats should be the same but are just specified in the logging file and are easy enough to change.

I'll link to where the changes are, but what specifically was making them hard to read?

danjampro, 2021-01-28:
Examples of logs using `tail`:

Normal `panoptes.log`:
```
W 01-27 23:58:35.988 (23:58:35 UTC) | panoptes.utils.config.client get_config:106 | Problem with get_config: ConnectionError(MaxRetryError("HTTPConnectionPool(host='localhost', port=6563): Max retries exceeded with url: /get-config (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fc5a4393250>: Failed to establish a new connection: [Errno 111] Connection refused'))"))
D 01-27 23:58:35.991 (23:58:35 UTC) | huntsman.pocs.archive.archiver __init__:51  | Archive directory: None
W 01-27 23:58:36.000 (23:58:36 UTC) | panoptes.utils.config.client get_config:106 | Problem with get_config: ConnectionError(MaxRetryError("HTTPConnectionPool(host='localhost', port=6563): Max retries exceeded with url: /get-config (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fc5a4398590>: Failed to establish a new connection: [Errno 111] Connection refused'))"))
```

Old/archived `panoptes_20210127.log`:
```
{"text": "2021-01-27 23:57:18.677 | WARNING  | panoptes.utils.config.client:get_config:106 - Problem with get_config: ConnectionError(MaxRetryError(\"HTTPConnectionPool(host='localhost', port=6563): Max retries exceeded with url: /get-config (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f52900cc5d0>: Failed to establish a new connection: [Errno 111] Connection refused'))\"))\n", "record": {"elapsed": {"repr": "0:00:00.406582", "seconds": 0.406582}, "exception": null, "extra": {"padding": ""}, "file": {"name": "client.py", "path": "/var/panoptes/panoptes-utils/src/panoptes/utils/config/client.py"}, "function": "get_config", "level": {"icon": "\u26a0\ufe0f", "name": "WARNING", "no": 30}, "line": 106, "message": "Problem with get_config: ConnectionError(MaxRetryError(\"HTTPConnectionPool(host='localhost', port=6563): Max retries exceeded with url: /get-config (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f52900cc5d0>: Failed to establish a new connection: [Errno 111] Connection refused'))\"))", "module": "client", "name": "panoptes.utils.config.client", "process": {"id": 1, "name": "MainProcess"}, "thread": {"id": 139993254086464, "name": "MainThread"}, "time": {"repr": "2021-01-27 23:57:18.677644+00:00", "timestamp": 1611791838.677644}}}
{"text": "2021-01-27 23:57:18.679 | TRACE    | panoptes.utils.config.client:get_config:119 - No config entry found, returning  default=None\n", "record": {"elapsed": {"repr": "0:00:00.408910", "seconds": 0.40891}, "exception": null, "extra": {}, "file": {"name": "client.py", "path": "/var/panoptes/panoptes-utils/src/panoptes/utils/config/client.py"}, "function": "get_config", "level": {"icon": "\u270f\ufe0f", "name": "TRACE", "no": 5}, "line": 119, "message": "No config entry found, returning  default=None", "module": "client", "name": "panoptes.utils.config.client", "process": {"id": 1, "name": "MainProcess"}, "thread": {"id": 139993254086464, "name": "MainThread"}, "time": {"repr": "2021-01-27 23:57:18.679972+00:00", "timestamp": 1611791838.679972}}}
{"text": "2021-01-27 23:57:18.680 | TRACE    | panoptes.utils.config.client:get_config:122 - Config key='archiver.delay_interval':  config_entry=None\n", "record": {"elapsed": {"repr": "0:00:00.409682", "seconds": 0.409682}, "exception": null, "extra": {}, "file": {"name": "client.py", "path": "/var/panoptes/panoptes-utils/src/panoptes/utils/config/client.py"}, "function": "get_config", "level": {"icon": "\u270f\ufe0f", "name": "TRACE", "no": 5}, "line": 122, "message": "Config key='archiver.delay_interval':  config_entry=None", "module": "client", "name": "panoptes.utils.config.client", "process": {"id": 1, "name": "MainProcess"}, "thread": {"id": 139993254086464, "name": "MainThread"}, "time": {"repr": "2021-01-27 23:57:18.680744+00:00", "timestamp": 1611791838.680744}}}
```

wtgee, 2021-01-29:
Ah yes, the archive files are stored in json format, which could be switched easily. The idea was to have some better log analysis program that you could use to search through them and those would prefer json, but since we are not doing that yet there is probably not a practical reason at this point to keep them in that format.

wtgee, 2021-04-10:
This option can be toggled with param added here: https://github.com/panoptes/POCS/pull/1094/files

closed 2021-05-17 by danjampro