AstroHuntsman/huntsman-pocs #303: SkyX "Another script is running" error

danjampro, 2021-01-13:
This error occurred after slewing to target for the first time after waiting for the observe horizon. It seems to have been caused by a call to `mount.get_status` during a slew, possibly by the monitoring thread in `POCS` or `Observatory` (can't remember which). Here is the error:
```S 01-13 10:48:15.013 Successful transition to slewing
I 01-13 10:48:15.073 Run loop: self.state='slewing'self.next_state='focusing'
I 01-13 10:48:15.751 Going to self.next_state='focusing'
C 01-13 10:48:21.185 Problem going from self.state='slewing' to  self.next_state='focusing', exiting loop [TheSkyXTimeout()]
S 01-13 10:48:21.190 Stopping POCS states
I 01-13 10:48:21.196 Calling set_config on  url='http://0.0.0.0:6563/set-config'
In [2]: E 01-13 10:49:17.249 TheSkyXError: Another socket command is still in progress. Error = 219
E 01-13 10:49:17.256 TheSkyXError:  Another script is running!
W 01-13 10:49:17.259 Can't get mount status: TheSkyXError(' Another script is running!')
```
The only solution is to restart both theSkyX and the state machine.

wtgee, 2021-01-13:
I believe in TSX you can turn off the JavaScript pop up errors that prevent further action.

danjampro, 2021-01-14:
Closed by https://github.com/panoptes/POCS/pull/1065

closed 2021-01-14 by danjampro