AstroHuntsman/huntsman-pocs #330: Increase BANDWIDTHOVERLOAD on RPI's to 100%

Physarah, 2021-01-21:
Currently all cameras have data transfer speed throttled back to 40% (and it resets every time the camera is power-cycled from jetson experiments). Lets try to increase this to 100% and see what happens? 

Do do this with something like:

```
camera = ASICamera(library_path='/home/nvidia/Tests/ASI_linux_mac_SDK_V1.16.3/lib/armv8/libASICamera2.so'
camera.set_control_value(0,'BANDWIDTHOVERLOAD', 100)
```

Physarah, 2021-01-21:
I wonder if this should be a PR to huntsman-pocs, or pocs itself?

danjampro, 2021-01-21:
I think we *should* be able to configure this using the config. I wonder if decreasing it to ~20% may also help. Either way, worth an experiment.

danjampro, 2021-01-27:
See https://github.com/panoptes/POCS/pull/1071