AstroHuntsman/huntsman-pocs #389: "sleep does not exist" state machine error

danjampro, 2021-03-04:
```
C 03-04 18:52:16.387 (18:52:16 UTC) | panoptes.pocs.state.machine run:141                                       | Problem going from 'twilight_flat_fielding' to 'parking', exiting loop [AttributeError("'sleep' does not exist on <Machine@140511380752128>")]
S 03-04 18:52:16.391 (18:52:16 UTC) | panoptes.pocs.state.machine stop_states:211                               | Stopping POCS states
```

wtgee, 2021-03-04:
I don't think this is a state error.

Our `pocs.sleep` has disappeared and this is an attribute error from this line:

https://github.com/AstroHuntsman/huntsman-pocs/blob/6209da6881abd39eab238773c57ff1a7990cb48c/src/huntsman/pocs/states/huntsman/twilight_flat_fielding.py#L16

wtgee, 2021-03-05:
I'm not sure why `pocs.sleep` isn't there though. I don't seem to recall that going away. 😕

wtgee, 2021-03-05:
Needs to be `pocs.wait` and was missed with the major docker update a few months ago.

danjampro, 2021-05-17:
Fixed

closed 2021-05-17 by danjampro