AstroHuntsman/huntsman-pocs #444: Use AAT weather

danjampro, 2021-05-27:


wtgee, 2021-05-27:
This gets most of the way there. I don't know which of the `unknown_` columns are the actual safety readings, but I'm sure someone does.

Just need to add a `safe: True/False` value based off those columns.

```
import requests

url = 'http://aat-ops.anu.edu.au/met/metdata.dat'

columns = [
    'time',
    'ambient_temp',
    'dome_temp',
    'mirror_temp',
    'outside_dewpoint',
    'outside_humidity',    
    'pressure',
    'wind_speed_avg',
    'wind_gust_max',
    'wind_direction_avg',
    'unknown_3',
    'unknown_4',
    'unknown_5',
    'sky_ambient_diff_C',
    'sky_ambient_diff_error',    
    'boltwood_wetness_meter_maybe',
    'unknown_7',
    'unknown_8',
    'rain_since_9am',
    'sqm_brightness',
]

res = requests.get(url)
if res.ok:
    date, raw_data, _ = res.content.decode().split('\n')
    data = {name:value for name, value in zip(columns, raw_data.split('\t'))}
    data['date'] = date
```

fergusL, 2021-06-03:
Here are the column names according to Steve Lee:

UT+10 i.e. local time, but NOT daylight saving time
Out_T Outside temperature in degrees C
In_T Inside dome temperature
Mir_T Temperature of the AAT primary mirror (measured a few millimetres above the actual surface)
Dew_O Dewpoint temperature of outside air
Hum_O% Humidity of outside air
Bar_mm Barometric pressure in mm of Mercury (for historical reasons)
Wkph Measured average wind speed in kph
WMax Maximum measured wind gust
WDir Direction of wind in degrees (0=north, 90=east, etc.)
D Flag for Dome state 0=Closed, 2=Venting, 4=Open
R Flag for Rain 0=no rain, 1=Rain
Dew_D Dewpoint temperature of dome air
Sky-Amb Boltwood reported sky-ambient temperature
+/-dev standard deviation of previous measurements
Bright Boltwood reported sky brightness
BR Boltwood rain sensor = number of times in 10 minute period rain detected
BW Boltwood wetness sensor (as above) : values 0, 1...10; -1 no reading
Rainfall Cumulative rainfall for day (reset at 9am local time).

closed 2021-07-28 by danjampro