AstroHuntsman/huntsman-pocs #496: Crash after restarting and seeing safety = False

Physarah, 2021-06-29:
Behaviour: 
Restarted state machine to test weather, Huntsman saw that it was unsafe to open and so closed the dome, but did not go into the taking darks state, it just crashes. 

Expected Behaviour: 
After determining it is unsafe to observe, close the dome and go into flats/darks state? 

D 06-29 13:18:52.769 Dark Check: True
D 06-29 13:18:52.861 Checking weather safety
D 06-29 13:18:53.005 Weather Safety: True [137 sec old - 2021-06-29 13:16:35]
W 06-29 13:18:53.296 Unsafe conditions: {'ac_power': True, 'is_dark': True, 'good_weather': False, 'free_space_root': True, 'free_space_images': True}
W 06-29 13:18:53.296 Unsafe conditions: {'ac_power': True, 'is_dark': True, 'good_weather': False, 'free_space_root': True, 'free_space_images': True}
C 06-29 13:18:53.306 Problem going from 'taking_darks' to 'starting', exiting loop [RuntimeError('Safety check failed!')]
C 06-29 13:18:53.306 Problem going from 'taking_darks' to 'starting', exiting loop [RuntimeError('Safety check failed!')]
I 06-29 13:18:53.312 Parking the telescope before stopping states.
I 06-29 13:18:53.312 Parking the telescope before stopping states.
D 06-29 13:18:53.317 Checking safety for <Transition('taking_darks', 'parking')@140142820831488>
D 06-29 13:18:53.319 Always safe to move to parking
S 06-29 13:18:53.322 Unit says: Entering starting state from the taking_darks state.
S 06-29 13:18:53.322 Unit says: Entering starting state from the taking_darks state.
D 06-29 13:18:53.519 Mount status: {'ra_rate': 15.0410644, 'dec_rate': 0, 'tracking': False, 'slewing': False, 'connected': True, 'error': 0, 'parked': True, 'ra': 7.649666606460772, 'dec': -20.4290722881962, 'alt': '-31.79', 'az': '210.98', 'ha': '10.1444', 'success': True}
D 06-29 13:18:53.774 Restarting Timer (blocking) 60.00/60.00
D 06-29 13:18:53.777 Sleeping for 60.00 seconds
D 06-29 13:18:54.269 State change status: {'state': 'parking', 'system': {'free_space': '47.877324800000004 Gbyte'}, 'observatory': {'can_observe': True, 'mount': {'tracking_rate': '1.0000', 'ra_guide_rate': 0.9, 'dec_guide_rate': 0.9, 'movement_speed': '', 'current_ra': <Longitude 114.74459956 deg>, 'current_dec': <Latitude -20.42907221 deg>, 'ra_rate': 15.0410644, 'dec_rate': 0, 'tracking': False, 'slewing': False, 'connected': True, 'error': 0, 'parked': True, 'ra': 7.649666606460772, 'dec': -20.4290722881962, 'alt': '-31.79', 'az': '210.98', 'ha': '10.1444', 'success': True, 'current_ha': <Longitude 10.12993744 hourangle>}, 'dome': {'dome_loop_running': True, 'status_loop_running': True, 'keep_shutter_open': True, 'Shutter': 'Open', 'Door': 'Closed', 'Battery': 12.6931, 'Solar_A': 0.0433594, 'Switch': 'EM243A'}, 'observer': {'siderealtime': '17h46m46.7987s', 'utctime': <Time object: scale='utc' format='datetime' value=2021-06-29 13:18:53.332825>, 'localtime': datetime.datetime(2021, 6, 29, 13, 18, 53, 570036), 'local_evening_astro_tiing_astro_time': <Time object: scale='utc' format='jd' value=2459395.31687596>, 'local_sun_set_time': <Time object: scale='utc' format='jd' value=2459394.796413067>, 'local_sun_rise_time': <Time object: scale='utc' format='jd' value=2459395.3799436465>, 'local_sun_position': <Latitude -76.67097573 deg>, 'local_moon_alt': <Latitude 15.97247204 deg>, 'local_moon_illumination': 0.728928657470875, 'local_moon_phase': <Quantity 1.09521278 rad>}}}
S 06-29 13:18:54.272 Unit says: Parking Huntsman.
S 06-29 13:18:54.272 Unit says: Parking Huntsman.
I 06-29 13:18:54.277 Setting new observation to None
I 06-29 13:18:54.277 Setting new observation to None
D 06-29 13:18:54.282 Already connected to dome
I 06-29 13:18:54.285 Closed dome
I 06-29 13:18:54.285 Closed dome
I 06-29 13:18:54.289 Closing dome shutter.
I 06-29 13:18:54.289 Closing dome shutter.

Physarah, 2021-06-29:
This only happens sometimes though, kind of odd. When I tried to reproduce the error it didn't occur

danjampro, 2021-07-28:
@Physarah this looks like normal behaviour to me. The state machine tried to leave the darks state to go into "starting", but the weather was not safe. In this case, an error gets thrown to immediately leave the state (in future this can be a `NotSafeError` or something. The point is that the exception itself is not a problem. 

It looks like it went into park right after this, which is intended behaviour. The question is - what happened next? Did it go into darks?