AstroHuntsman/huntsman-pocs #536: Closed loop slew functionality

Physarah, 2021-09-08:
This is just a wish list item for me. Enabling closed loop slew before we start an observation would ensure we are always on target.

danjampro, 2021-09-08:
What is the main barrier in getting this done? The interface between `POCS` and TSX?

Physarah, 2021-09-08:
I believe so. The trigger from POCS to initiate the closed loop slew (with guide cam presumably), then ensuring POCS doesn't move the mount again afterwards.

danjampro, 2021-09-08:
Am I correct in thinking this is this something that is already implemented in TSX GUI?

Physarah, 2021-09-08:
yup - so it's just a matter of triggering this from POCS some how

danjampro, 2021-09-08:
I think it would require a new item in `POCS` here:

https://github.com/panoptes/POCS/tree/develop/resources/bisque/mount

The only problem is that I have no idea how to write these things. I think @wtgee wrote them originally.

@wtgee, do you have any pointers on how to do this?